udacity/robotics-nanodegree-issues #173: pcl_to_ros() and/or ros_to_pcl() methods in pcl_helper.py (for project # 3) are messing up the point cloud

safdark, 2017-11-07:
I have been able to reproduce this on my machine, using the file attached.

In project # 3, we are to write the pick+place pipeline for object recognition for PR2_robot. In this pipeline, pcl_helper.py is available for us to use. I was having issues with the point cloud I would produce at the end of this pipeline, but it turns out that the issue might be in pcl_to_ros() or ros_to_pcl(). See attached screenshot.

safdark, 2017-11-07:
[image]

safdark, 2017-11-07:
[Uploading Screen Shot 2017-11-06 at 1.38.29 PM.png...]

safdark, 2017-11-07:
Notice in the first attachment, that I am using the PointCloud2 visualizwr to view the /pcl_raw topic. I have subscribedd to the /camera/depth_registered/points topic, and done the following:

# Callback function for your Point Cloud Subscriber
def pcl_callback(pcl_msg):
    # TODO: Convert ROS msg to PCL data
    pcl_raw = ros_to_pcl(pcl_msg)
    
    # TODO: Convert PCL data to ROS messages
    ros_raw = pcl_to_ros(pcl_raw)

    # TODO: Publish ROS messages
    pcl_original_pub.publish(pcl_msg)
    pcl_raw_pub.publish(ros_raw)

if __name__ == '__main__':

    # TODO: ROS node initialization
    rospy.init_node("advanced_pick_and_place")

    # TODO: Create Subscribers
    pcl_sub = rospy.Subscriber("/camera/depth_registered/points", PointCloud2, pcl_callback, queue_size=1)

    # TODO: Create Publishers
    pcl_original_pub = rospy.Publisher("/pcl_original", PointCloud2, queue_size=1)
    pcl_raw_pub = rospy.Publisher("/pcl_raw", PointCloud2, queue_size=1)

    # TODO: Spin while node is not shutdown
    while not rospy.is_shutdown():
        rospy.spin()    
Ideally, this should be the same as the original pcl_msg object received

safdark, 2017-11-07:
Here's teh full source code of the reproducible issue:

orientation_problem.py:
---------------------------------

#!/usr/bin/env python

# Import modules
import numpy as np
import sklearn
from sklearn.preprocessing import LabelEncoder
import pickle
from sensor_stick.srv import GetNormals
from sensor_stick.features import compute_color_histograms
from sensor_stick.features import compute_normal_histograms
from visualization_msgs.msg import Marker
from sensor_stick.marker_tools import *
from sensor_stick.msg import DetectedObjectsArray
from sensor_stick.msg import DetectedObject
from sensor_stick.pcl_helper import *

import rospy
import tf
from geometry_msgs.msg import Pose
from std_msgs.msg import Float64
from std_msgs.msg import Int32
from std_msgs.msg import String
from pr2_robot.srv import *
from rospy_message_converter import message_converter
import yaml

# Helper function to get surface normals
def get_normals(cloud):
    get_normals_prox = rospy.ServiceProxy('/feature_extractor/get_normals', GetNormals)
    return get_normals_prox(cloud).cluster

# Helper function to create a yaml friendly dictionary from ROS messages
def make_yaml_dict(test_scene_num, arm_name, object_name, pick_pose, place_pose):
    yaml_dict = {}
    yaml_dict["test_scene_num"] = test_scene_num.data
    yaml_dict["arm_name"]  = arm_name.data
    yaml_dict["object_name"] = object_name.data
    yaml_dict["pick_pose"] = message_converter.convert_ros_message_to_dictionary(pick_pose)
    yaml_dict["place_pose"] = message_converter.convert_ros_message_to_dictionary(place_pose)
    return yaml_dict

# Helper function to output to yaml file
def send_to_yaml(yaml_filename, dict_list):
    data_dict = {"object_list": dict_list}
    with open(yaml_filename, 'w') as outfile:
        yaml.dump(data_dict, outfile, default_flow_style=False)

# Callback function for your Point Cloud Subscriber
def pcl_callback(pcl_msg):

# Exercise-2 TODOs:

    # TODO: Convert ROS msg to PCL data
    pcl_raw = ros_to_pcl(pcl_msg)
    
#    # TODO: Convert PCL data to ROS messages
    ros_raw = pcl_to_ros(pcl_raw)

#    # TODO: Publish ROS messages
    pcl_original_pub.publish(pcl_msg)
    pcl_raw_pub.publish(ros_raw)

if __name__ == '__main__':

    # TODO: ROS node initialization
    rospy.init_node("advanced_pick_and_place")

    # TODO: Create Subscribers
    pcl_sub = rospy.Subscriber("/camera/depth_registered/points", PointCloud2, pcl_callback, queue_size=1)

    # TODO: Create Publishers
    pcl_original_pub = rospy.Publisher("/pcl_original", PointCloud2, queue_size=1)
    pcl_raw_pub = rospy.Publisher("/pcl_raw", PointCloud2, queue_size=1)

    # TODO: Spin while node is not shutdown
    while not rospy.is_shutdown():
        rospy.spin()

safdark, 2017-11-07:
```
#!/usr/bin/env python

# Import modules
import numpy as np
import sklearn
from sklearn.preprocessing import LabelEncoder
import pickle
from sensor_stick.srv import GetNormals
from sensor_stick.features import compute_color_histograms
from sensor_stick.features import compute_normal_histograms
from visualization_msgs.msg import Marker
from sensor_stick.marker_tools import *
from sensor_stick.msg import DetectedObjectsArray
from sensor_stick.msg import DetectedObject
from sensor_stick.pcl_helper import *

import rospy
import tf
from geometry_msgs.msg import Pose
from std_msgs.msg import Float64
from std_msgs.msg import Int32
from std_msgs.msg import String
from pr2_robot.srv import *
from rospy_message_converter import message_converter
import yaml

# Helper function to get surface normals
def get_normals(cloud):
    get_normals_prox = rospy.ServiceProxy('/feature_extractor/get_normals', GetNormals)
    return get_normals_prox(cloud).cluster

# Helper function to create a yaml friendly dictionary from ROS messages
def make_yaml_dict(test_scene_num, arm_name, object_name, pick_pose, place_pose):
    yaml_dict = {}
    yaml_dict["test_scene_num"] = test_scene_num.data
    yaml_dict["arm_name"]  = arm_name.data
    yaml_dict["object_name"] = object_name.data
    yaml_dict["pick_pose"] = message_converter.convert_ros_message_to_dictionary(pick_pose)
    yaml_dict["place_pose"] = message_converter.convert_ros_message_to_dictionary(place_pose)
    return yaml_dict

# Helper function to output to yaml file
def send_to_yaml(yaml_filename, dict_list):
    data_dict = {"object_list": dict_list}
    with open(yaml_filename, 'w') as outfile:
        yaml.dump(data_dict, outfile, default_flow_style=False)

# Callback function for your Point Cloud Subscriber
def pcl_callback(pcl_msg):

# Exercise-2 TODOs:

    # TODO: Convert ROS msg to PCL data
    pcl_raw = ros_to_pcl(pcl_msg)
    
#    # TODO: Convert PCL data to ROS messages
    ros_raw = pcl_to_ros(pcl_raw)

#    # TODO: Publish ROS messages
    pcl_original_pub.publish(pcl_msg)
    pcl_raw_pub.publish(ros_raw)

if __name__ == '__main__':

    # TODO: ROS node initialization
    rospy.init_node("advanced_pick_and_place")

    # TODO: Create Subscribers
    pcl_sub = rospy.Subscriber("/camera/depth_registered/points", PointCloud2, pcl_callback, queue_size=1)

    # TODO: Create Publishers
    pcl_original_pub = rospy.Publisher("/pcl_original", PointCloud2, queue_size=1)
    pcl_raw_pub = rospy.Publisher("/pcl_raw", PointCloud2, queue_size=1)

    # TODO: Spin while node is not shutdown
    while not rospy.is_shutdown():
        rospy.spin()
```

safdark, 2017-11-07:
This is blocking my progress on project # 3, because I need these methods to work on the image pipeline.

To reproduce this issue:
1) Drop this code (above) into the ~/catkin_ws/src/RoboND-Perception-Project/scripts/ folder.
2) Run: 
       roslaunch pr2_robot pick_place_project.launch
3) Run: 
      ./orientation_problem.py

safdark, 2017-11-07:
[image]

safdark, 2017-11-07:
Note in this last image above, that I have displayed 2 topics:
/pcl_original <-- The original message
/pcl_raw <-- The same message after going through ros_to_pcl() and pcl_to_ros()

/pcl_original displays the flat table (in its right location, which is expected since this is the message received on the callback function that I am just echoing to the /pcl_original topic)

/pcl_raw displays the tilted/displaced version, which got messed up through the two helper invocations above.

safdark, 2017-11-09:
[image]

This (attached) is what I see when I run "roswtf" in my catkin workspace.

sahiljuneja, 2017-11-09:
Hey! Can you share your github repo with the latest code here? I will try to run it locally and see what the issue might be. Haven't seen this before so wish to cross-check. I will try to get back to you asap accordingly!

safdark, 2017-11-09:
Hi Sahil,
Sure. Here's the github link for the code I wrote to reproduce this (since
it didn't require the entire pipeline). You can check the other files too,
if you'd like.

https://github.com/safdark/ROBO-perception-project/blob/master/RoboND-Perception-Project/pr2_robot/scripts/orientation_issue.py

Hope this helps?

safdark, 2017-11-09:
> master/RoboND-Perception-Project/pr2_robot/scripts/orientation_issue.py

Also, note that the pcl_helper.py file exists in two locations:
pr2_robot/scripts/pcl_helper.py
sensor_stick/src/sensor_stick/pcl_helper.py

They're both the same file actually (just as it was in the Udacity repos).
I didn't check which of the two of those actually got imported, but it
didn't matter since the files were identical.

safdark, 2017-11-10:
I have logged another, possibly related, issue (
https://github.com/udacity/robotics-nanodegree-issues/issues/176), but this
other issue involves loss of color information (no orientation change). It
appears that the color information is being lost via a call to
ros_to_pcl(). I have provided the instructions to reproduce the issue.
Thought I'd post a comment about it here since it might be simple to
reproduce the 2nd issue in the same environment needed as the current issue.

sahiljuneja, 2017-11-10:
The issue seems to stem from how the Subscriber is defined in your code. 

Because of that a wrong transformation is being applied (somewhere, I am not sure where exactly and will have to look more into it but it's most likely how the t f tree is setup for the topic you are trying to subscribe to) but that's why one of the topics is publishing it at a different transformation than expected.

I would recommend following the lesson closely to avoid similar problems.

https://classroom.udacity.com/nanodegrees/nd209/parts/586e8e81-fc68-4f71-9cab-98ccd4766cfe/modules/e5bfcfbd-3f7d-43fe-8248-0c65d910345a/lessons/e3e5fd8e-2f76-4169-a5bc-5a128d380155/concepts/0f453188-f83b-48fd-b0d3-49f3d5bfed55

```
First, you'll need to change your subscriber (or create a new subscriber) to subscribe to the camera data (point cloud) topic /pr2/world/points.
```

I tried with the correct Subscriber and it's publishing to both the topics correctly.

closed 2017-11-27 by sahiljuneja

safdark, 2017-11-28:
Thanks Sahil! Your comment about how to fix the issue worked like a charm.