udacity/robotics-nanodegree-issues #176: ros_to_pcl() seems to be causing a loss of color information

safdark, 2017-11-10:
I am trying to capture ROS point clouds and save them to disk, for which I need to convert the sensor_msg/PointCloud2 type to the pcl PointCloud_ type. After conversion, I save the point cloud to disk using pcl.save() from the pcl. However, when I view the saved point clouds, the color information of each object is largely missing. (See attachments)

safdark, 2017-11-10:
[image]

safdark, 2017-11-10:
[image]

Both these attachments are showing the side of the objects that is facing the camera (not the rear side).

safdark, 2017-11-10:
```
To reproduce this:

1) mkdir -p ~/catkin_ws/src; cd ~/catkin_ws/src
2) git clone https://github.com/safdark/ROBO-perception-project ./RoboND-Perception-Project
3) roslaunch sensor_stick training.launch
4) roscd sensor_stick/scripts
5) ./capture_point_clouds.py -y ../config/pick_list_0.yaml -c 5 -o ./clouds
6) cd clouds
7) Subfolders convey the model name, and under each folder are some point clouds. Use 'pcl_viewer' to view them.
```

Note: The _capture_point_clouds.py_ file is just capturing the point cloud from the requisite ROS topic, and is then invoking *_ros_to_pcl()_* on it. That is where, it appears, this issue loss of color information might be happening.

safdark, 2017-11-10:
[image]

Here's the same loss of color information for the otherwise *very* colorful beer can.

safdark, 2017-11-10:
I'm unable to upload the .pcd file type, otherwise I would have attached the corresponding file for this beer point cloud.

danzelmo, 2017-11-17:
Hi safdark, 

I am unsure what is causing the issue here. I can't say when but this is something I would like to look into. Are pointclouds in general supposed to contain color information, if it is generated from lidar I thought color info was lost. It is something I would like to know if it is a bug or not, but it might be a while to look into/potentially fix if issue is found.

yycho0108, 2018-04-07:
I was also noticing the same thing, and I believe it is actually an issue with capture_sample()

[image]

When the point cloud is loaded, it appears to undergo a brief period where the colors are grayed out for some reason. In my case, I added a 1 sec. delay until I actually captured the data. This obviously made the data collection process super long, but I didn't really have any other option.

What's really surprising to me is the fact that this issue was not resolved for a full 5 months until now.