udacity/robotics-nanodegree-issues #184: Cannot use GPU on udacity robotics AMI on AWS

sandyleo26, 2017-11-23:
For follow me project, I followed the instruction to create p2.xlarge instance and installed the RoboND env but training is slow (800s /epoch). GPU memory usage is 0. It's because tensorflow-gpu is not used. But it cannot be installed correctly because no cudnn driver. I have to create a new instance using amazon's own deep learning ubuntu AMI and create the RoboND there. Then it works and 1 epoch takes only a minute.

sahiljuneja, 2017-11-23:
Hey!

For the AMI you don't have to install the RoboND env. You can work directly in the AMI which comes preinstalled with tensorflow-gpu. 

The RoboND env will install tensorflow-cpu instead.

I just checked the AMI again, and everything works well there. I will close the issue for now. Good luck!

closed 2017-11-23 by sahiljuneja

sandyleo26, 2017-11-23:
I see. Well then maybe the instructions in the classroom could be clearer coz I follow it and source robond.

sahiljuneja, 2017-11-24:
Can you clarify where you noticed some ambiguity?

Because the current section on working with AWS in the Project lesson doesn't ask you to install the RoboND environment in it.

sandyleo26, 2017-11-24:
You’re right the instructions is correct. I guess it’s switching back and forth between aws and local environment that trip me over. I don’t know about others but it’ll definitely help if it’s noted that one uses RoboND while the others doesn’t. Or a RoboND env is created on aws rather than using the root env.