udacity/robotics-nanodegree-issues #185: Deadline Surpassage

pravbeatle, 2017-11-23:
Even though I was locked out of the content due to not completing all projects before the required deadline, I was still able to bypass that blockage by clicking on the 'Resume Learning' button in Classroom home.

pravbeatle, 2017-11-23:
[image]
[image]

jchernus, 2018-02-08:
Thanks! Udacity is aware of this issue, and a team is working on fixing this.

closed 2018-02-08 by jchernus