udacity/robotics-nanodegree-issues #187: Frequent "Process has died" errors in vanilla environment on Udacity VM

safdark, 2017-11-30:
I’m trying to understand what causes this error below. This error seems to indicate that the environment has gotten corrupted, because often if I restart the VM and clean out the build/ and devel/ folders, and do a fresh “catkin_make”, this issue resolves itself (without any changes to the code). However, this happens frequently enough, it appears, to warrant investigation by the Udacity team.

It is unfortunately the most frustrating part of working with the ROS modules in this course, to have to deal with this instability. If this were to be resolved, it would save an enormous amount of time that is otherwise now spent troubleshooting course material that fails independent of students' code.

What is the cause of this, and why is ROS so unstable?Also, why does it reference a log file that never exists? Is there a cleaner way of working around this bug?

```
[setup_2d_world-1] process has died [pid 75556, exit code 2, cmd /home/robond/catkin_ws/src/RoboND-Controls-Lab/quad_controller/scripts/setup_2d_hover_world.sh __name:=setup_2d_world __log:=/home/robond/.ros/log/2fd6164c-d5d8-11e7-86e9-000c2939d11d/setup_2d_world-1.log].
log file: /home/robond/.ros/log/2fd6164c-d5d8-11e7-86e9-000c2939d11d/setup_2d_world-1*.log
```

safdark, 2017-11-30:
I don't have a reproducible scenario for this, but it happens often enough to be rather frustrating. Most recently, I am seeing this error when I try to launch:

> roslaunch quad-controller hover_controller_node

safdark, 2017-11-30:
The solution is to employ a _combination_ of 2 or more of the below:
1) Clean out the /build and /devel folders and run catkin_make
2) Clean out the src/RoboND-Controls-Lab folder entirely, and reclone, and run catkin_make
3) Restart the VM

#2 and #3 seem to be required more often than not. #1 almost rarely works on its own.

kylesf, 2018-03-12:
Moving to workspaces. VM will be deprecated. Thanks for submitting!

closed 2018-03-12 by kylesf