udacity/robotics-nanodegree-issues #192: Can't follow links in class notes

jazzdman, 2017-12-10:
In TensorFlow for Deep Learning, section 15: TensorFlow Cross Entropy, there are two links for multi-class cross entropy and One Hot Encoding.     Also section 30: Lab: TensorFlow Neural Networks has a link for instructions on how to setup TensorFlow, I believe.  When I click on any of those links, my browser (Safari) simply directs me back to the main syllabus page.

jazzdman, 2017-12-11:
The link for classroom setup in section 6 Lab: Getting Started of Lesson 32:  Semantic Segmentation.  Neither does the link that asks you to "Click here" for package requirements.

jazzdman, 2017-12-13:
For me, this problem appears to be complete (i.e any link that I click on sends me back to the syllabus).  I have noticed this also occurs in the Chrome browser in the RoboND VM.

sahiljuneja, 2017-12-13:
Thanks for pointing this out. Some recent update might be causing this issue. I will start looking into this.

sahiljuneja, 2017-12-13:
Most, if not all, such broken links should be fixed now. If you encounter more, that I might have missed, then please create a new ticket. Thanks for pointing these out!

closed 2017-12-13 by sahiljuneja