udacity/robotics-nanodegree-issues #193: No module named 'moviepy' in online rover lab

Kevin-La-Ra, 2017-12-20:


Kevin-La-Ra, 2017-12-20:
In code section 11, "Make a video from processed image data"

# Import everything needed to edit/save/watch video clips
from moviepy.editor import VideoFileClip

results in 

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-13e3c6f195dc> in <module>()
      1 # Import everything needed to edit/save/watch video clips
----> 2 from moviepy.editor import VideoFileClip
      3 from moviepy.editor import ImageSequenceClip
      4 
      5 

ModuleNotFoundError: No module named 'moviepy'

Adding "import moviepy" does not fix the issue.

kylesf, 2018-01-19:
Fixed should be deployed. Sorry about any inconvenience!

closed 2018-04-26 by sahiljuneja