udacity/robotics-nanodegree-issues #194: Lesson 11.7: Composition of rotations multiplies rotations in same direction.

satsugaisha, 2017-12-27:
In the above mentioned lesson it asks us to code both an intrinsic and an extrinsic rotation, with the difference between these two being the rotation order. However the solutions for both of these code quizzes has the same multiplication order.

satsugaisha, 2017-12-27:
This his means that the answers to the two quizzes are different, making the answer to the third quiz, a multiple choice question, incorrect.

jchernus, 2018-02-08:
I am seeing the solution to both as being,

YZ_intrinsic_sym = R_y * R_z

Is this the case for you?

satsugaisha, 2018-02-10:
Yes, it's the same for me, and as I understand it, the order of multiplication should be different for extrinsic and intrinsic.

sahiljuneja, 2018-05-28:
A late response, so I'm hoping you got this clarified somewhere else. But regardless -

The text above the quiz mentions 

> The first requires you to perform an intrinsic rotation sequence about the Y and then Z axes. The second quiz requires you to perform an extrinsic rotation about the Z and then Y axes.

As per the video, intrinsic rotations are post-multiplied. So , the order of multiplication would be Ry*Rz for the first one (because first rotation is about Y).

Extrinsic rotations are pre-multiplied. So, order of multiplication would be Ry*Rz (because first rotation is about Z).

So, yes. Both rotations in both quizzes are the same.

closed 2018-05-28 by sahiljuneja