udacity/robotics-nanodegree-issues #197: `an_image.list` does not contain valid filepaths

flores-jacob, 2018-01-18:


flores-jacob, 2018-01-18:
Under Term 2 Lesson 6 part 4 - Image classification with DIGITS - improving performance, right after the text:
```
We can classify multiple files if we put them in the list. In the link below, execute the code block and a link to the file an_image.list will appear. Right click on an_image.list and save that to a file on your local computer(right click and "Save As"). Remember the directory in which it is saved.
```
where the student is asked to save a `an_image.list` to use to test the trained models, the `an_image.list` contains paths that do not map to actual files, resulting in the errors below when trying to use them to check the trained models.

```
Tensorflow support disabled.
"/notebooks/test_images/image-1-1.jpg" not found
"/notebooks/test_images/image-2-1.jpg" not found
"/notebooks/test_images/image-3-1.jpg" not found
"/notebooks/test_images/image-4-1.jpg" not found
"/notebooks/test_images/image-7-1.jpg" not found
"/notebooks/test_images/image-8-1.jpg" not found
"/notebooks/test_images/image-8-2.jpg" not found
2018-01-18 09:26:33 [ERROR] InferenceError: Unable to load any image from file ''/home/workspace/digits_jobs/20180118-092631-7b00/tmp0kXXhv.txt''
Traceback (most recent call last):
File "/opt/DIGITS/digits/tools/inference.py", line 293, in <module>
args['resize']
File "/opt/DIGITS/digits/tools/inference.py", line 151, in infer
raise InferenceError("Unable to load any image from file '%s'" % repr(input_list))
digits.inference.errors.InferenceError: "Unable to load any image from file ''/home/workspace/digits_jobs/20180118-092631-7b00/tmp0kXXhv.txt''"
```

Contents of the `an_image.list` file:

```
/notebooks/test_images/image-1-1.jpg
/notebooks/test_images/image-2-1.jpg
/notebooks/test_images/image-3-1.jpg
/notebooks/test_images/image-4-1.jpg
/notebooks/test_images/image-7-1.jpg
/notebooks/test_images/image-8-1.jpg
/notebooks/test_images/image-8-2.jpg
```

priteshgudge, 2018-01-19:
Same issue for me too.
[image]

kylesf, 2018-01-19:
Hi there, I have sent out for new build of image to include the data. For now please enter this command into a terminal on your digits workspace `curl -L https://s3-us-west-1.amazonaws.com/udacity-robotics/Content+Workspace+Use/lab_data.tar.gz |  tar zx -C /`

closed 2018-05-03 by sahiljuneja