udacity/robotics-nanodegree-issues #214: EKF lab launch file issue

adammalpass, 2018-02-19:
I found an error in the packing naming used inside the main launch file. I made a pull request here:
https://github.com/udacity/RoboND-EKFLab/pull/1

adammalpass, 2018-02-20:
As part of the lab students do not directly use the RoboND-EKF repo. Instead the robot_pose_ekf repo is cloned and the other packages made up during the lab. Only for the main launch file itself does the lab direct to refer to the RoboND-EKF repo instead (as an aside, this is confusing and I only noticed this discrepancy after reading your comment and checking into further - perhaps these can be merged into one repo?).

Furthermore, if you check the RVIZ lesson (part 8) in this section it specifically states to save the .rviz file at this location:
~/catkin_ws/src/EKFLab.rviz

Either way, something is wrong/misleading.

karimchamaa, 2018-02-21:
Yeah, you are right, the file should be under src directly as I asked in Concept n8. Thanks for reporting the issue.

closed 2018-02-21 by karimchamaa