udacity/robotics-nanodegree-issues #215: Inverse Sensor Model Slide

streklin, 2018-02-22:


karimchamaa, 2018-02-22:
Hey, can you provide more details about the issue?

streklin, 2018-02-22:
This is a bit of a nitpick, but the letter spacing on the Inverse Sensor model slide should be adjusted.  As it is the slide is very difficult to read correctly as it seems to read "ifr" instead of "if r" (see image).  While this in no way degrades ones ability  to complete the course, it is confusing when reading over the slide until whats really meant "clicks" in.
[image]

If someone can find a way to simply increase that spacing a bit it would vastly improve the readibility of the slide.  Again, this really is a bit of a nitpick on my part, but its a simple fix that may reduce confusion for others.

karimchamaa, 2018-02-22:
I agree, thanks for reporting this issue. We are currently fixing that and should replace this video soon with the newer version.

closed 2018-06-05 by karimchamaa