udacity/robotics-nanodegree-issues #216: Where am I Project BUG

chedanix, 2018-02-22:
The friction coefficients are not set properly for the casters because SDF format was used for defining the friction. But take a look at udacity_bot.xacro, the <SDF> tag is not declared. As a result the caster wheels still have a non-zero default friction coefficient. This leads to bumpy movement or rover not moving in a straight line. I have done experiments on my model, and so I am pretty certain about this.

chedanix, 2018-02-22:
I found a way to fix it.
1) remove front and back casters from the "chassis" link.
2) define two links: front caster and back caster
3) define two joints from chassis to casters
4) in udacity_bot.gazebo, add following for each caster
 <gazebo reference="xxx_caster">
   <mu1>0.0</mu1>
   <mu2>0.0</mu2>
 </gazebo>

You might be wondering. Can we just add the following to udacity_bot.gazebo instead?
 <gazebo reference="chassis">
   <mu1>0.0</mu1>
   <mu2>0.0</mu2>
 </gazebo>
It didn't work for me. You can try though.

sahiljuneja, 2018-02-22:
Thanks for reporting this. I will cross-check on this soon and update accordingly!