udacity/robotics-nanodegree-issues #217: Where am I Project - Map flickering

chedanix, 2018-02-23:
I believe the root cause of map flickering comes from static_transform_publisher in amcl.launch.

Most people, myself included initially, simply tune out the map flickering by increasing the transform_tolerance  parameter under amcl. This parameter post-dates a transform into the future, increasing its validity. Therefore, using it to "cover up" the transform from static_transform_publisher may not be wise.

static_transform_publisher serves no purpose and should be removed.

sahiljuneja, 2018-02-23:
Yes, as I pointed in the Slack, this might be the cause. Will test this out soon before updating the classroom. Thanks for creating the ticket!

sahiljuneja, 2018-02-25:
Fixed!

closed 2018-02-25 by sahiljuneja