udacity/robotics-nanodegree-issues #219: Lesson 17 - Grid-based FastSLAM

Things2Connect, 2018-02-27:
I have problems completing sections 17.13 & 17.14 (gmapping with ROS).  I have used the turtlebot_gazebo and turtlebot_teleop packages from lesson 11 (Lab: Kalman Filters). I have completed the necessary steps: 1) Cloned the gmapping package 2) Installed dependencies and built my catkin workspace 3) Launched the nodes of the 3 different packages 4) Run and edited RViz 5) Teleoperated the robot. However, I have not been able to complete the last step: obtaining a map of the environment.  When I run command: 

roslaunch turtlebot_gazebo turtlebot_world.launch world_file:=worlds/willowgarage.world

The command launches the Willow Garage world in Gazebo,  however, I get following error in the terminal:

ERROR] [1519755884.101047072, 0.166000000]: Failed to load nodelet [/depthimage_to_laserscan] of type [depthimage_to_laserscan/DepthImageToLaserScanNodelet] even after refreshing the cache: Failed to load library /opt/ros/kinetic/lib//libDepthImageToLaserScanNodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_core3.so.3.3: cannot open shared object file: No such file or directory)
[ERROR] [1519755884.101230638, 0.166000000]: The error before refreshing the cache was: Failed to load library /opt/ros/kinetic/lib//libDepthImageToLaserScanNodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_core3.so.3.3: cannot open shared object file: No such file or directory)
[FATAL] [1519755884.117705152, 0.170000000]: Failed to load nodelet '/depthimage_to_laserscan` of type `depthimage_to_laserscan/DepthImageToLaserScanNodelet` to manager `laserscan_nodelet_manager'

I am not sure what this error means ?

The nodes for the other two packages (teleop and gmapping) can be launched and seem to be running OK.  I am using VM image: VM2.1.0 (VM image from lesson 5 back in Term 1).

karimchamaa, 2018-02-27:
I am not sure why you are having that. Seems like a node failed to launch. Did you run an update and upgrade? And did you ever install the OpenCV library yourself on this VM?

Things2Connect, 2018-02-28:
I used the VM image provided in Term 1 (VM version 2.1.0 provided in lesson 5 - "Introduction to ROS").  I haven't run any updates or upgrades myself apart from what's instructed in Lessons 11  & 17 in Term 2  (Lesson 11: "Lab: Kalman Filters" and Lesson 17: ""Grid-Based FastSLAM").    Also, I haven't installed openCV on the VM image myself.   The openCV version on the VM image would be what was originally installed in VM V2.10.  So, I'm not sure why the error is happening.  I went onto Slack yesterday, and I have noticed that other students in the class have come across a similar error.  But I'm not sure how to solve it.

karimchamaa, 2018-02-28:
Still not sure why you are facing this problem but run these commands and see if your problem is solved:

sudo apt-get update
sudo apt-get upgrade
sudo apt-get update
cd ~/catkin_ws
source devel/setup.bash
rosdep update 
rosdep -i install turtlebot_gazebo 
catkin_make
source devel/setup.bash
roslaunch turtlebot_gazebo turtlebot_world.launch world_file:=worlds/willowgarage.world

If this doesn't work for you then I advise you to try it out in the workspace especially that you'll have to use it for your project soon.

closed 2018-03-09 by karimchamaa