udacity/robotics-nanodegree-issues #220: Lesson 13.2: Links to papers from RSEND page are broken

PaulNWms, 2018-03-01:
It looks like 2 of the papers linked to from the RSEND page are no longer available. (?)  At least I am not able to open them:
All  robots are not created equal: The design and perception of humanoid robot heads
By Carl F. Disalvo et. al
Robot  ethics: a view from the philosophy of science
By Guglielmo Tamburrini

https://sites.google.com/knowlabs.com/udacity-robotics-slack/udacity-explores/human-robot-interaction

I also left a message about this in the #hri channel

PaulNWms, 2018-03-01:
The link in Lesson 13.4 is also broken.  (Disalvo paper)
But the link in Lesson 13.7 works. (Tamburrini paper)

haoyang-f, 2018-03-20:
Fixed, thank you for the input!

closed 2018-03-20 by haoyang-f