udacity/robotics-nanodegree-issues #223: Project - Map My World Robot - Project Setup for Jetson TX2

Things2Connect, 2018-03-05:
I would like to use my Jetson TX2 for the project "Map My World Robot".  I have flashed the board and downloaded JetPack 3.1, as instructed at the beginning of Term 2.  However, I am unable to install ROS and RTAB, as per instructions for this project (section 12 in the project description).  I have followed the instructions step by step, but I encounter errros when I follow these instructions. 

To be specific there are two issues during the installation on my Jetson:

1) When I carry out the first step (Install ROS), it runs for about 25 mins.  Everything seems ok 
until it gets towards the end of the installation and gives an error saying "cannot download default source list https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down"

I was not sure what to do at this point, but I carried on as this happened towards the end of installation.

2)  When I got to the third step (Install RTAB-MAP), the system halted at 82% giving the error: "recipe for target 'all' failed"  and stopped.    

I have provided the relevant output from the terminal on my Jetson TX2:
.
.
.
Setting up ros-kinetic-gazebo-dev (2.5.14-1xenial-20171221-164251-0800) ...
Setting up ros-kinetic-gazebo-plugins (2.5.14-1xenial-20180125-233236-0800) ...
Setting up ros-kinetic-gazebo-ros (2.5.14-1xenial-20180126-015737-0800) ...
Setting up ros-kinetic-gazebo-ros-pkgs (2.5.14-1xenial-20180126-041209-0800) ...
Setting up ros-kinetic-simulators (1.3.1-0xenial-20180221-102826-0800) ...
Setting up ros-kinetic-desktop-full (1.3.1-0xenial-20180221-104719-0800) ...
Processing triggers for fontconfig (2.11.94-0ubuntu1) ...
Processing triggers for initramfs-tools (0.122ubuntu8) ...
Processing triggers for libc-bin (2.23-0ubuntu3) ...
Processing triggers for systemd (229-4ubuntu6) ...
ERROR: cannot download default sources list from:
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down.
..
..
..
[ 82%] Linking CXX executable ../../../bin/rtabmap
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmMap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
app/src/CMakeFiles/rtabmap.dir/build.make:468: recipe for target '../bin/rtabmap' failed
make[2]: *** [../bin/rtabmap] Error 1
CMakeFiles/Makefile2:416: recipe for target 'app/src/CMakeFiles/rtabmap.dir/all' failed
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

Things2Connect, 2018-03-05:
I've attached the entire log for my installation as well.

[ROS_installation_Jetxon_TX2.log](https://github.com/udacity/robotics-nanodegree-issues/files/1781733/ROS_installation_Jetxon_TX2.log)

Things2Connect, 2018-03-05:
I've attached the entire logfile for the project setup as well:

[Logfile_ROS_installation_Jetxon_TX2.txt](https://github.com/udacity/robotics-nanodegree-issues/files/1781737/Logfile_ROS_installation_Jetxon_TX2.txt)

Things2Connect, 2018-03-05:
I've attached the entire logfile for the installation as well:

[Logfile_ROS_installation_Jetxon_TX2.txt](https://github.com/udacity/robotics-nanodegree-issues/files/1781742/Logfile_ROS_installation_Jetxon_TX2.txt)

Things2Connect, 2018-03-09:
I've managed to solve these two problems now, and I have completed the installation of ROS & RTAB-Map on my Jetson TX2.  

Issue 1:  
---------

The fault message looked like this:

**"cannot download default source list https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down"**  

This fault prevented Rosdep from working when I tried to run the simulation:

**^Cnvidia@tegra-ubuntu:~$ sudo rosdep init
[sudo] password for nvidia: 
Sorry, try again.
[sudo] password for nvidia: 
ERROR: cannot download default sources list from:
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down.
nvidia@tegra-ubuntu:~$** 

Without Rosdep working,  I could not install the dependencies for rtabmap_ros hence making it impossible to run the simulation.  

However, I found a solution to this on ros.org:

**https://answers.ros.org/question/54150/rosdep-initialization-error/**

See below for my implemenation of this:

**nvidia@tegra-ubuntu:~$ rosdep check --from-paths catkin_ws/src

ERROR: your rosdep installation has not been initialized yet.  Please run:

    sudo rosdep init
    rosdep update

nvidia@tegra-ubuntu:~$ sudo rosdep init
[sudo] password for nvidia: 
ERROR: cannot download default sources list from:
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down.
nvidia@tegra-ubuntu:~$ sudo c_rehash /etc/ssl/certs
Doing /etc/ssl/certs
WARNING: Skipping duplicate certificate ca-certificates.crt
WARNING: Skipping duplicate certificate ca-certificates.crt
nvidia@tegra-ubuntu:~$ sudo rosdep init
Wrote /etc/ros/rosdep/sources.list.d/20-default.list
Recommended: please run

	rosdep update

nvidia@tegra-ubuntu:~$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"
Add distro "hydro"
Add distro "indigo"
Add distro "jade"
Add distro "kinetic"
Add distro "lunar"
Add distro "melodic"
updated cache in /home/nvidia/.ros/rosdep/sources.cache
nvidia@tegra-ubuntu:~$** 

Issue 2:
-----------

The error looked like this:

**CMakeFiles/Makefile2:416: recipe for target 'app/src/CMakeFiles/rtabmap.dir/all' failed
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2**

I solved it by removing the RTAB-map directory on my Jetson and applying this solution provided in the project description:

**cd /usr/lib/aarch64-linux-gnu && sudo rm libGL.so && sudo ln -s /usr/lib/aarch64-linux-gnu/tegra/libGL.so libGL.so**

I then re-ran following commands from the project description and the RTAB-Map installation was successful:

**cd ~ && git clone https://github.com/introlab/rtabmap.git rtabmap && cd rtabmap/build && cmake .. && make && sudo make install** 

So, in the end, I managed to completed the installation of ROS & RTAB-Map on my Jetson TX2.

karimchamaa, 2018-04-26:
It seems like you didn't install rosdep when installing ROS.

closed 2018-05-02 by karimchamaa