udacity/robotics-nanodegree-issues #224: Lesson 16 (11. Inverse Sensor Model)

chedanix, 2018-03-08:
Alpha shown in the picture and video is not correct. According to Probabilistic Robotics by Thrun, alpha should be the thickness of obstacles.

Also, please add somewhere in the lecture that  a sonar rangefinder is used for the Inverse Sensor Model. It confused me for a long time because I thought a laser rangefinder was used and wondered where the "measurement cone" came from.

karimchamaa, 2018-03-09:
Thanks for reporting the issues and sorry for the confusion. It's all being fixed in the classroom, please take a look.

closed 2018-03-09 by karimchamaa

chedanix, 2018-03-09:
I just took a look. It hasn't been fixed yet, right?

karimchamaa, 2018-03-09:
The picture/video hasn't been fixed, just added the description - summary and mentioned that it's a sonar rangefinder sensor