udacity/robotics-nanodegree-issues #225: RTAB-Map does not work on my Jetson TX2

Things2Connect, 2018-03-09:
When I launch the file mapping.launch on my Jetson TX2, the rtab-map process keeps on dying in my terminal.  

However, if I run the same robot package on my workspace, I do not get this error and RTAB-Map starts up without failure.  So, it seems to have something to do with my Jetson TX2.  I have followed the instructions for installing ROS & RTAB-Map on my TX2, as per project description.  There were a few problems but I managed to sort them out.  See my previous waffle ticket #223.

This is the error I get on my Jetson TX2:

[ INFO] [1520537762.383313964, 216.408000000]: rtabmap 0.15.0 started...
[rtabmap/rtabmap-1] process has died [pid 21274, exit code -11, cmd /home/nvidia/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start scan:=/scan rgb/image:=/camera/rgb/image_raw depth/image:=/camera/depth/image_raw rgb/camera_info:=/camera/rgb/camera_info grid_map:=/map __name:=rtabmap __log:=/home/nvidia/.ros/log/5726740e-2307-11e8-addc-00044ba7dade/rtabmap-rtabmap-1.log].
log file: /home/nvidia/.ros/log/5726740e-2307-11e8-addc-00044ba7dade/rtabmap-rtabmap-1*.log

Things2Connect, 2018-03-11:
Also, further comments about this issue. 

1) My Jetson TX2 is flashed with Jetpack 3.1 done via an Ubuntu 14.04 host (I folllowed the instructions at the beginning of Term 2)

2)  When I run file mapping.launch on my Jetson TX2 and then in my workspace following difference appear on TX2:

 **WARN] (2018-03-11 17:07:19.432) Features2d.cpp:407::create() SURF and SIFT features cannot be used because OpenCV was not built with nonfree module. ORB is used instead.**

The entire output for roslaunch on my Jetson TX2 is shown below:

~/catkin_ws$ roslaunch slam_project mapping.launch
... logging to /home/nvidia/.ros/log/a8aae338-2545-11e8-a33e-00044ba7dade/roslaunch-tegra-ubuntu-3859.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:35999/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /rtabmap/rtabmap/Kp/DetectorStrategy: 0
 * /rtabmap/rtabmap/Kp/MaxFeatures: 400
 * /rtabmap/rtabmap/Mem/NotLinkedNodesKept: false
 * /rtabmap/rtabmap/Reg/Force3DoF: true
 * /rtabmap/rtabmap/Reg/Strategy: 0
 * /rtabmap/rtabmap/Rtabmap/DetectionRate: 1
 * /rtabmap/rtabmap/SURF/HessianThreshold: 100
 * /rtabmap/rtabmap/Vis/MinInliers: 15
 * /rtabmap/rtabmap/database_path: rtabmap.db
 * /rtabmap/rtabmap/frame_id: base_footprint
 * /rtabmap/rtabmap/odom_frame_id: odom
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_scan: True

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)

ROS_MASTER_URI=http://localhost:11311

process[rtabmap/rtabmap-1]: started with pid [3876]
[ INFO] [1520784432.337446153]: Starting node...
[ INFO] [1520784432.423948294]: Initializing nodelet with 4 worker threads.
[ INFO] [1520784434.264759828, 225.874000000]: /rtabmap/rtabmap(maps): map_filter_radius          = 0.000000
[ INFO] [1520784434.264856786, 225.874000000]: /rtabmap/rtabmap(maps): map_filter_angle           = 30.000000
[ INFO] [1520784434.264897745, 225.874000000]: /rtabmap/rtabmap(maps): map_cleanup                = true
[ INFO] [1520784434.264930289, 225.874000000]: /rtabmap/rtabmap(maps): map_negative_poses_ignored = false
[ INFO] [1520784434.264969904, 225.874000000]: /rtabmap/rtabmap(maps): map_negative_scan_ray_tracing = true
[ INFO] [1520784434.265003023, 225.874000000]: /rtabmap/rtabmap(maps): cloud_output_voxelized     = true
[ INFO] [1520784434.265034510, 225.874000000]: /rtabmap/rtabmap(maps): cloud_subtract_filtering   = false
[ INFO] [1520784434.265072173, 225.874000000]: /rtabmap/rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[ INFO] [1520784434.364509463, 225.959000000]: rtabmap: frame_id      = base_footprint
[ INFO] [1520784434.364624501, 225.959000000]: rtabmap: odom_frame_id = odom
[ INFO] [1520784434.364673780, 225.959000000]: rtabmap: map_frame_id  = map
[ INFO] [1520784434.364714163, 225.959000000]: rtabmap: tf_delay      = 0.050000
[ INFO] [1520784434.364756114, 225.959000000]: rtabmap: tf_tolerance  = 0.100000
[ INFO] [1520784434.364794417, 225.959000000]: rtabmap: odom_sensor_sync   = false
[ INFO] [1520784435.365933047, 226.899000000]: Setting RTAB-Map parameter "Kp/DetectorStrategy"="0"
[ INFO] [1520784435.421528888, 226.951000000]: Setting RTAB-Map parameter "Kp/MaxFeatures"="400"
[ INFO] [1520784435.696097478, 227.216000000]: Setting RTAB-Map parameter "Mem/NotLinkedNodesKept"="false"
[ INFO] [1520784437.007561058, 228.465000000]: Setting RTAB-Map parameter "Reg/Force3DoF"="true"
[ INFO] [1520784437.010070284, 228.467000000]: Setting RTAB-Map parameter "Reg/Strategy"="0"
[ INFO] [1520784437.061742972, 228.510000000]: Setting RTAB-Map parameter "Rtabmap/DetectionRate"="1"
[ INFO] [1520784437.323787772, 228.757000000]: Setting RTAB-Map parameter "SURF/HessianThreshold"="100"
[ INFO] [1520784437.780202083, 229.188000000]: Setting RTAB-Map parameter "Vis/MinInliers"="15"
[ WARN] [1520784438.307738597, 229.691000000]: Setting "Grid/FromDepth" parameter to false (default true) as "subscribe_scan" or "subscribe_scan_cloud" is true. The occupancy grid map will be constructed from laser scans. To get occupancy grid map from cloud projection, set "Grid/FromDepth" to true. To suppress this warning, add <param name="Grid/FromDepth" type="string" value="false"/>
[... 1 lines cut ...]
[ INFO] [1520784439.431809064, 230.761000000]: rtabmap: Deleted database "/home/nvidia/.ros/rtabmap.db" (--delete_db_on_start or -d are set).
[ INFO] [1520784439.431933126, 230.761000000]: rtabmap: Using database from "/home/nvidia/.ros/rtabmap.db" (0 MB).
[ WARN] (2018-03-11 17:07:19.432) Features2d.cpp:407::create() SURF and SIFT features cannot be used because OpenCV was not built with nonfree module. ORB is used instead.
[ WARN] (2018-03-11 17:07:19.436) Features2d.cpp:407::create() SURF and SIFT features cannot be used because OpenCV was not built with nonfree module. ORB is used instead.
[ WARN] (2018-03-11 17:07:19.441) Features2d.cpp:407::create() SURF and SIFT features cannot be used because OpenCV was not built with nonfree module. ORB is used instead.
[ WARN] (2018-03-11 17:07:19.564) Features2d.cpp:407::create() SURF and SIFT features cannot be used because OpenCV was not built with nonfree module. ORB is used instead.
[ INFO] [1520784439.567984876, 230.893000000]: rtabmap: Database version = "0.15.0".
[ INFO] [1520784439.720931569, 231.035000000]: /rtabmap/rtabmap: queue_size    = 10
[... cut ...]
[rtabmap/rtabmap-1] process has died [pid 3876, exit code -11, cmd /home/nvidia/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start scan:=/scan rgb/image:=/camera/rgb/image_raw depth/image:=/camera/depth/image_raw rgb/camera_info:=/camera/rgb/camera_info grid_map:=/map __name:=rtabmap __log:=/home/nvidia/.ros/log/a8aae338-2545-11e8-a33e-00044ba7dade/rtabmap-rtabmap-1.log].
log file: /home/nvidia/.ros/log/a8aae338-2545-11e8-a33e-00044ba7dade/rtabmap-rtabmap-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
~/catkin_ws$

karimchamaa, 2018-04-26:
Are you still getting errors?

closed 2018-05-02 by karimchamaa