udacity/robotics-nanodegree-issues #234: Term 1, 6.2.2: Active Sensors Typo

delvingdeep, 2018-03-29:
There seems to be minor typo in Concept 6: Perception, Module 2: Introduction to 3D Perception and Lesson 2: `Active sensor`.

The first table is about difference between Active and Passive sensors; but the Table heading are does not match. 

The Heading should be `Passive Sensors` instead of `Passive Sensing` or the either way `Active 
Sensing` and not `Active Sensors`.

delvingdeep, 2018-03-29:
[image]

sahiljuneja, 2018-05-25:
Fixed. Thanks!

closed 2018-05-25 by sahiljuneja