udacity/robotics-nanodegree-issues #237: Term 1, Lesson 24: Simulator not responding to cmd_force messages

sonelu, 2018-04-13:


sonelu, 2018-04-13:
Hello,

I am experiencing some errors with the quad simulator for the lab in lesson 24 of term 1 (Quadrotor Control Using PID).

My environment is MacOS 10.13.04 (High Siera) and I run the VM in VirtualBox using a clean installation of Ubuntu 16.04 and ROS. I have been using this setting for Project 1 and 2 of the term without any problems.

The VM is setup in Bridge mode for network so that the guest OS has it's own IP address in addition to the IP address of the host OS. Once VM is started I can ping the host OS from the VM and the guest OS from MacOS.

I start the environment as follows:
- `roscore` on VM
- simulator in MacOS - this indicates it is connected to ROS (I have maintained the roc_settings.txt as indicated)
- `roslaunch quad_controller hover_controller.launch` to start the hover node

The PID code is implemented and if I adjust the `target` and the `kp` parameters in the rqt_reconfigure widow messages are sent. I can see that if I run `rostopic echo -c /quad_rotor/cmd_force`:

`force:
  x: 0.0
  y: 0.0
  z: 9.91749992222
torque: 
  x: 0.0
  y: 0.0
  z: 0.0`

(for a 10m target and 1 for kp). 

The quad in the simulator doesn't do anything.

I can see that the nodes are properly connected to the topics:

`alex@ros:~$ rostopic info /quad_rotor/cmd_force `
`Type: geometry_msgs/Wrench`

`Publishers: `
` * /hover_controller (http://ros:42505/)`

`Subscribers: `
` * / (http://Alex2.local:50444/)`

And I can also send commands to the quad in the simulator by invoking `rosservice`:

`alex@ros:~$ rosservice call /quad_rotor/set_pose "pose:`
`  position:`
`    x: 0.0`
`    y: 0.0`
`    z: 20.0`
`  orientation:`
`    x: 0.0`
`    y: 0.0`
`    z: 0.0`
`    w: 0.0"`
`success: True`
`message: ''`

I can see the quad bouncing up in the simulator and then getting back to the ground under gravity.

In VM I can also run `rostopic echo -c /quad_rotor/pose` and I can see the topic being published by the simulator. If I activate input on in the simulator and manually change the position of the quad (ex. I press space bar to increase Z position) I can see the position correctly being published and visible in the VM.

But all messages /quad_rotor/cmd_force seem to be ignored by the simulator.

sonelu, 2018-04-13:
I have to add that the behaviour is the same (no moving) for any values of the message values (ex. with Kp set at 100 and requesting force Z of 1000.0). In all cases the values reported in the simulator window for the force are all 0.0:
[image]

sonelu, 2018-04-18:
Gents, is anyone looking at these issues? 

Thanks,

Alex.

jessluls, 2018-05-03:
We're looking to move this into an online platform in the classroom so this shouldn't be an issue once that happens. Thank you for flagging this.

closed 2018-05-03 by jessluls