udacity/robotics-nanodegree-issues #244: Term 2, Robotics Software Engineer Nanodegree Program, Lesson 27: Lab: Path Planning

douglasteeple, 2018-04-29:
The output window in this whole lesson on MacOSX Sierra, Safari, shows a very narrow and unreadable x size. I have a screen shot but don't know how to attach it...

douglasteeple, 2018-04-29:
[image]

karimchamaa, 2018-04-30:
[image]

karimchamaa, 2018-04-30:
I just tried it on a MacOSX Sierra Safari browser and it's working fine! Maybe deleting cache and cookies will solve your problem.

closed 2018-05-02 by karimchamaa